ramiss/arduino_DJI_03_RC_ARM #2: O3 input levels are 3v3

Kisssys, 2023-02-04:
I believe the Nano has 5 volt TTL and I believe the O3 shows 3v3 volt levels.  Maybe the inside is 5 volt tolerant but makes me nervous
to try this.
Cheers
Kisssys

ramiss, 2023-02-28:
You could use 1k resistors on the signal wires to step down the voltage

Kisssys, 2023-02-28:
Yes, you could, I just programmed a Seeduino XIAO board which is 3v3 volts
logic.
1/3 the size.
Cheers
Kisssys

Slapjack30, 2023-06-16:
Hi @Kisssys,

Did you need to change a lot for the code, using the XIAO?

Kisssys, 2023-06-16:
The code is attached for a XIAO Sam21.

I believe how the serial ports were defined was the main change.

Cheers
Kisssys

Slapjack30, 2023-06-16:
thanks! 

and did you make the voltage divider too? on this XIAO? which resistors did you use to get it down to 3.3v?

TIA

Kisssys, 2023-06-16:
That processor operates on 3v3 volts, so it's output is fine the way it is.

Slapjack30, 2023-06-16:
i mean the voltage divider coming from Vbat (20V) to not 5V as in the example from Ramiss, but we need to get the 20V down to 3.3V for voltage info..

Kisssys, 2023-06-16:
I saw no reason to read the voltage since the DJI displays it already.  I
removed that code.  You could add it and use a 39k/7.5k resistors for a max
of 20 volt would get you below 3.3 volts for the A2D
converter.

Slapjack30, 2023-06-17:
Thanks a lot!!!

Slapjack30, 2023-06-24:
could you share your code Kisssys? the one supplied doenst work with the XIAO...

Kisssys, 2023-06-25:
The code was attached previously.

Slapjack30, 2023-06-25:
i cant see your code in the project, could you be so kind to copy past your code? or send it to me? your code, if you have send yours to the owner, hasnt been updates yet in the master code...

TIA

ramiss, 2023-06-25:
I've merged main into the master branch so you should see it now

Slapjack30, 2023-06-26:
Awesome guys!!! thank a lot for your work! i hope to see the arm channel included in the XIAO version, because fixed wing flyers will power up the aircraft, do the pre flight checks, which can be a couple of minutes, and then take off... so plenty of time for the O3 unit to heat up, possibly causing problems. so arm channel would be great, but so far... awesome project!

cheers!

ramiss, 2023-06-26:
I hope to get it added in the next week or so.  The big hurdle is that the XIAO is 3.3v and RC receiver PWM logic is 5v, so a logic level converter will be needed. Its more work than I had last weekend and I didn't want to wait to publish yesterday.

closed 2023-09-08 by ramiss