ramiss/arduino_DJI_03_RC_ARM #4: Wont arm

mikkik64, 2023-04-24:
Hi. I have set the Arduino up as per your instructions and it doers not appear to arm. I am using this on a RC buggy. I plugged the arm channel into channel 3 of the RC buggy receiver and arm the O3 and switch channel 3 on when the buggy is in motion. Is that right or have I got that bit wrong? Thanks

ramiss, 2023-04-24:
There are quite a few reasons this might not be working, so it's a bit impossible to answer from your post.  It's the sketch actually uploaded,  is the wiring exactly correct,  are you connecting the goggles to the air unit prior to arming,  is your channel 3 outputting the correct values (ie not reversed,  etc)?

mikkik64, 2023-04-24:
Sketch is uploaded (I think, I'll double check), wiring is correct with 5v UBEC and connecting goggles to O3 before arming. Good point re C3 being reversed. I'll plug it it C4 as my RC C4 switch can go both ways so I ill test it with it plugged into C4. Thanks

ramiss, 2023-05-29:
Did you get this resolved?

closed 2023-07-30 by ramiss