ramiss/arduino_DJI_03_RC_ARM #5: How to keep unit always armed

Beento, 2023-05-29:
I'm using the O3 in a rocket to record video but need a way to keep it always armed whenever powering on, Is there a simple way to modify the code to do so or should it be a hardware solution?

ramiss, 2023-05-29:
Yes, its possible.  I'll make an update and get back to you if you wouldn't mind testing.

ramiss, 2023-05-29:
> I'm using the O3 in a rocket to record video but need a way to keep it always armed whenever powering on, Is there a simple way to modify the code to do so or should it be a hardware solution?

I've added a setting (line 22) which you can set to true and see if this works for you.  Please let me know.

Beento, 2023-05-29:
Thank you! Also, How do you recommend to power the Air unit and Arduino? I was planning on wiring them both to a single 2s lioni battery but seems like you need to power the air unit first. Any thoughts?

ramiss, 2023-05-29:
> Thank you! Also, How do you recommend to power the Air unit and Arduino? I was planning on wiring them both to a single 2s lioni battery but seems like you need to power the air unit first. Any thoughts?

Follow the wiring diagram and you should be good.

ramiss, 2023-05-29:
> Thank you! Also, How do you recommend to power the Air unit and Arduino? I was planning on wiring them both to a single 2s lioni battery but seems like you need to power the air unit first. Any thoughts?

Sorry, for got the "no ESC" part of rockets.. lol.   Yes, wire them both to the battery.  One does not need to boot up before the other.

Beento, 2023-05-29:
Sounds great thank you. Is a 2s lion enough voltage for both?

ramiss, 2023-05-29:
> Sounds great thank you. Is a 2s lion enough voltage for both?

Apologies.  the Arduino Nano only runs on 5v, which means you'll need a uBEC or generic 5v regulator of some kind.  They are pretty cheap.  Search uBEC on Amazon or wherever.  

The 03 air unit will run on a 2S battery, but that is its lowest, so make sure you have a good charge.  you might try a low capacity 3S and see if you get better efficiency.

ramiss, 2023-05-29:
Also make sure you add the resistors between the Arduino and the Air Unit signal wires.

kilrah, 2023-05-29:
For the record I tested with a regulated supply and the O3 unit is happy down to a bit below 6V, so 2S is fine.

Beento, 2023-05-29:
> For the record I tested with a regulated supply and the O3 unit is happy down to a bit below 6V, so 2S is fine.

Was it at max power for the entire time? We need maybe 4-7 minutes for launch and after that its ok to die afterwards. 
And if so, could I probably power the Arduino off the same battery?

kilrah, 2023-05-29:
I got 25 mins recording with a 2s 700mAh battery at full TX power. You can power both with one battery if you connect the battery supply to the arduino's VIN pin to use the onboard regulator.

Beento, 2023-05-29:
Oh shoot, so if you wire the 2s voltage to the Vin pin, it bypasses needing a BEC? The Arduino will automatically figure that out?

kilrah, 2023-05-29:
The nano is specced 7-12V on VIN, it should be fine for 2-3s, You might want to check it does hold down to 6V but it most likely will. For other boards check the board's specs.

ramiss, 2023-05-29:
> I got 25 mins recording with a 2s 700mAh battery at full TX power. At 2s you can power both with one battery if you connect the battery supply to the arduino's VIN pin to use the onboard regulator.

Good catch!  Yes Nano VIN will work direct to battery.  All Arduinos are different though.

Beento, 2023-05-29:
Thank you! I'll try to use that then. Thanks guys!

ramiss, 2023-06-25:
The reason this wasn't auto arming was my misunderstanding.  Apparently the Air Unit must see an unarmed state before it sees an arm command.  I have reworked and tested the entire project for the much smaller XIAO Seeeduino, which is inherently 3.3v and compatible with the Air Unit. It will now auto arm. See Readme.

closed 2023-07-30 by ramiss