ramiss/arduino_DJI_03_RC_ARM #9: My o3 isnt arming

JasonITB, 2023-06-13:
I don't have a esc, throttle, or receiver connected. Only the o3 and the battery. I set bool always arm to true. Here's my wiring
[image](https://media.discordapp.net/attachments/1056499348776689686/1118088426940551188/IMG_3621.jpg?width=1003&height=753)

JasonITB, 2023-06-13:
I realized that last image was bad here's a better version. My o3 firmware is V01.02.0000 if that helps. Could I please get some support on this? Thank you
[image](https://media.discordapp.net/attachments/1056499348776689686/1118089063522648104/IMG_3622.jpg?width=564&height=753)

ramiss, 2023-06-25:
I have reworked and tested the entire project for the much smaller XIAO Seeeduino, which is inherently 3.3v and compatible with the Air Unit.  It will now auto arm.  See Readme.

closed 2023-06-26 by ramiss