ramiss/arduino_DJI_03_RC_ARM #13: XIAO ESP32C3 Error when flashing

PPPRY, 2023-07-15:
Compilation error: 'LED_BUILTIN' was not declared in this scope, First time writing the code to the board. I had no issue writing it to a nano board a few months ago. New to this, can I get a little help, please?!

ramiss, 2023-07-15:
The XIAO ESP32C3 is s different board to the original XIAO Seeeduino and will need the code modified.  You can try commenting out the LED lines as they aren't critical.

PPPRY, 2023-07-15:
[image]
So I switched the board over to seeedstudio. This is what I got https://www.amazon.com/Seeeduino-Smallest-Microcontroller-Interfaces-Compatible/dp/B08745JBRP/ref=sr_1_3?crid=1CD5QESFC58XJ&keywords=seeeduino+xiao&qid=1689388043&sprefix=seee%2Caps%2C176&sr=8-3

ramiss, 2023-07-15:
You misunderstand. The board needs to match. The code is not designed for the board you have.

PPPRY, 2023-07-15:
> The XIAO ESP32C3 is s different board to the original XIAO Seeeduino and will need the code modified. You can try commenting out the LED lines as they aren't critical.
https://www.seeedstudio.com/Seeeduino-XIAO-3Pcs-p-4546.html This is the original correct?

ramiss, 2023-07-15:
Correct,  with the SAMD21

PPPRY, 2023-07-15:
> Correct, with the SAMD21

So the last error was what I got after switching the board configuration over. The screen shot above. No longer the led error.

ramiss, 2023-07-15:
Did you add the Seeed Studio XIAO SAMD21 board in Arduino IDE? 

https://wiki.seeedstudio.com/Seeeduino-XIAO/

PPPRY, 2023-07-15:
Yes, bottom left of the screen shot above has the correct board?  #error MCU has no such a register
[image]

ramiss, 2023-07-15:
I wonder if you are not on the latest version of this code as it was significantly overhauled and the prior version was build for the Nano. After that,  make sure you are on the latest version of the Arduino IDE as well as making sure your libraries are up to date.

closed 2023-09-08 by ramiss