ramiss/arduino_DJI_03_RC_ARM #15: Compilation Error

arubano, 2023-08-19:
I'm a little new to this but followed the instructions carefully. I setup my Samd21 XIAO in IDE, it's recognized and I tested it.
I downloaded all of the files in this repository, opened them in the latest version of IDE and get an error.

I attached a txt file of the output errors. 

[Errors.txt](https://github.com/ramiss/arduino_DJI_03_RC_ARM/files/12386185/Errors.txt)

There are many errors, I'm sure I'm doing something wrong and I'm not sure what I'm looking at. Most of them are "stray" errors and telling me I can't perform certain actions at this time.

example.
error: stray '\342' in program
     You can’t perform that action at this time.
error: stray '\200' in program
     You can’t perform that action at this time.
error: stray '\231' in program
     You can’t perform that action at this time.
error: expected unqualified-id before '<' token
 <!DOCTYPE html>

ramiss, 2023-08-19:
I'm suspecting you downloaded the webpage instead of the code files.  The way you download code in GitHub is in the screenshot below. 

Once you unzip, place every file, except for the readme.md file in a folder called "arduino_DJI_03_RC_ARM"  (I'll fix the necessity for this step in the next update).

[image]

arubano, 2023-08-19:
Yes! lol I did a right click/save link as for each file because I didn't see a download button.

So I downloaded the zip, created the correct folder name, put the other files in it. Opened the sketch and tried to compile it but  I get an error about serialusb. All of the other errors are gone. I appreciate your response, your time and and any help you're willing to give me on this.

C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino: In function 'void setup()':
C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:81:5: error: 'SerialUSB' was not declared in this scope
     SerialUSB.begin(115200);
     ^~~~~~~~~
C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:81:5: note: suggested alternative: 'Serial'
     SerialUSB.begin(115200);
     ^~~~~~~~~
     Serial
C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino: In function 'void debugPrint()':
C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:334:3: error: 'SerialUSB' was not declared in this scope
   SerialUSB.println("**********************************");
   ^~~~~~~~~
C:\Users\antho\OneDrive\Desktop\Vers2\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:334:3: note: suggested alternative: 'Serial'
   SerialUSB.println("**********************************");
   ^~~~~~~~~
   Serial

exit status 1

Compilation error: 'SerialUSB' was not declared in this scope

ramiss, 2023-08-19:
Either you aren't using the latest version of Arduino IDE, or you aren't setting the correct board, or you some of your libraries are out of date.

ramiss, 2023-08-19:
[image]

arubano, 2023-08-19:
Okay. I will go through it all again. I'm using the Seeeduino SAMD21 XIAO https://www.amazon.com/dp/B0BQ39R32D

Arduino IDE is 2.1.1

In board manager I added the Seeed SAMD Boards from this link  https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json and it's on version 1.8.4

Board is set as Seeeduino XIAO on the correct port.

I wrote the blink test to it and changed the timings a few times on the blink to confirm it was writing and accepting changes.

I will go back through now and confirm all is correct.

arubano, 2023-08-19:
Mine looks a little different.

[image]

ramiss, 2023-08-19:
Hmm.   Did you follow the directions on this page (near the bottom)
https://wiki.seeedstudio.com/Seeeduino-XIAO/

ramiss, 2023-08-19:
The board looks correct.

arubano, 2023-08-19:
> Hmm. Did you follow the directions on this page (near the bottom) https://wiki.seeedstudio.com/Seeeduino-XIAO/

Yes, that is the page and the instructions I used to set it up.

arubano, 2023-08-19:
I downgraded from the 1.8.4 update in the board manager to 1.7.9 for this board and it compiled/uploaded successfully.

ramiss, 2023-08-19:
Interesting, thanks for the info and glad you got it working.

arubano, 2023-08-19:
An interesting power note. I expected gray/white from the air unit to just have data. This is in an r/c truck. I have it wired per your diagram. 

The Air Unit, Air Unit Fan and Seeeduino all get power from their own second battery in the truck. I have the power for the Seeeduino run through a mini buck converter and set for 5.00v exactly. Air Unit white is going to 6, gray is going to 7, brown is going to ground (shared with power ground). Power/Ground from buck converter are going to 12/13 with 5v on the 5v pin.

Before I even powered the seeeduino on, I powered on the air unit and the seeeduino came on by itself and worked. I tested with a volt meter, the power pin had 2.2 volts on it. I'd imagine the seeeduino or the air unit will get damaged working too long like this, so I connected power to it and had 5v where I was supposed to.. But it does appear work without applying any power to it.

Is there typically an indication from the Air Unit or in the goggles when it's entered high power transmit mode? Everything appears to look the same, range seems to be a little better, It still drops signal and video a couple hundred feet away if I go behind an obstacle or building. I also lose bandwidth pretty quickly too. A couple hundred feed out and I'm under 25 and I'm around 0-5 with 1 bar of signal at about half mile. I'm just trying to make sure I've done everything right. I can also see battery voltage in the goggles with the seeeduino connected and I don't remember seeing battery voltage before. 

Thanks again for your help and responding today, I do appreciate it.

ramiss, 2023-08-19:
Yeah, I'm also aware that the DJI UART alone will power the XIAO, but I agree its risky to rely on it.   If you are seeing anywhere near 25mbps or higher then it is armed.  You can also set it to auto record and that would give you indication.

arubano, 2023-08-19:
There are 3 modes from what I understand. There's low power. Then low power disabled from the goggles which increases bandwidth but keeps RF power low. Then full power bandwidth and full power RF with the arduino.

When I turn off low power in the goggles without the arduino I get 25-50mbps driving around no problem, but the range is only 300-400 feet. With the arduino in there, it doesn't seem much better.

I also turned on auto-recording which I assume is "Auto-Record on Takeoff" which does nothing. It doesn't start auto recording.

ramiss, 2023-08-19:
If it isn't auto recording then you aren't arming.  Are you sure RX and TX are appropriately reversed?

arubano, 2023-08-19:
I'm not sure if they're reversed or not. I have white (3rd wire) going to 6, gray (4th wire) going to 7 and the brown (5th wire) going to ground. Exactly as it is here.

[image]

ramiss, 2023-08-19:
If it's exact wiring then it should work.  You didn't change the code, I assume?

arubano, 2023-08-19:
I haven't. Aside from downgrading the board manager in IDE, everything was done how it was laid out in your instructions.

ramiss, 2023-08-19:
What is your startup process? The goggles need to be powered up before the air unit and arduino

ramiss, 2023-08-19:
You can uncomment the debug definition on line 34 in the main sketch, re-upload then then open the serial monitor to watch the logs on your pc as you power up.

arubano, 2023-08-19:
I power up goggles, then power up camera. Camera, fan and arduino are all on the same connector and connect to their own battery. The arduino starts as soon as I power up the air unit, even if I disconnect power to it so I’m not sure if the power up timing matters or not for that.

ramiss, 2023-08-19:
The Arduino waits for activity from the air unit to indicate it is booted up.  It then waits 3 seconds and sends the arm command.  I recommend doing the debug per above to see what is going on.

TodWulff, 2023-08-22:
I too was getting the serialusb error:
- Seeeds' boards added per guidance. 
- Latest IDE being used.

My fix:
- I commented out the serialusb lines inside of the debug function and then it compiled and flashed successfully.  Subsequently it's working as expected.

Not only jumps out of low power mode at boot, but also auto starts a recording.  I've some tuning to do with the 3S1P setup but otherwise it is a blessing.  Kudos and Thank You!

closed 2023-09-08 by ramiss