ramiss/arduino_DJI_03_RC_ARM #16: Error in compilation

3dstaff, 2023-08-29:
Hello, i try to upload the code to my original seeeduino samd21 but i get his errors in the image below, can you help please, my config seems ok.
[https://imageupload.io/rRyl5FyDearOQlO](url)

ramiss, 2023-08-29:
> Hello, i try to compile the code for my original seeeduino samd21 but i get this errors in the image below, can you help please, my config seems ok. It's a matter of compiling, the device works with other sketches. [image]

Make sure you downloaded all code files.

3dstaff, 2023-08-29:
Solved compilation error, forgot to add the .cpp file, now i have an upload error: "SAM-BA operation failed"

3dstaff, 2023-08-29:
Write 36528 bytes to flash (571 pages)

[===                           ] 11% (64/571 pages)
[======                        ] 22% (128/571 pages)
[==========                    ] 33% (192/571 pages)
[=============                 ] 44% (256/571 pages)
[================              ] 56% (320/571 pages)
[====================          ] 67% (384/571 pages)
[=======================       ] 78% (448/571 pages)
[==========================    ] 89% (512/571 pages)
[==============================] 100% (571/571 pages)
SAM-BA operation failed

done in 0.251 seconds

Verify 36528 bytes of flash with checksum.
Verify successful
done in 0.065 seconds
CPU reset.
Failed uploading: uploading error: exit status 1

3dstaff, 2023-08-29:
All solved code uploaded, need solder now :), thanks for this!

closed 2023-08-29 by 3dstaff