ramiss/arduino_DJI_03_RC_ARM #18: error message

rcflyer69, 2023-09-08:
i can't get this to work i need help

Desktop/arduino_DJI_03_RC_ARM-main/arduino_DJI_03_RC_ARM/arduino_DJI_03_RC_ARM.ino:224:35: warning: attribute ignored in declaration of 'struct msp_servo_mix_rules_t::<unnamed>' [-Wattributes]
  224 |   __attribute__ ((packed)) struct {
Desktop/arduino_DJI_03_RC_ARM-main/arduino_DJI_03_RC_ARM/arduino_DJI_03_RC_ARM.ino:224:35: note: attribute for 'struct msp_servo_mix_rules_t::<unnamed>' must follow the 'struct' keyword

ramiss, 2023-09-08:
Seems like you haven't downloaded all of the source files:
[image]

rcflyer69, 2023-09-09:
now I'm getting this issue 

/Desktop/arduino_DJI_03_RC_ARM-main/arduino_DJI_03_RC_ARM/arduino_DJI_03_RC_ARM.ino:334:3: error: 'SerialUSB' was not declared in this scope
   SerialUSB.println("**********************************");

ramiss, 2023-09-09:
Downgrade your board manager version to 1.7.9.  They changed something with SerialUSB that breaks the code after that.

closed 2023-09-10 by rcflyer69