ramiss/arduino_DJI_03_RC_ARM #22: Question:  OSD working on DJI 03 Air Unit?

Qrome, 2023-12-27:
Hey, just a quick question, I was wondering if you were able to get the OSD elements to display in the goggles using the DJI 03 Air Unit.   I don't currently have the 03 unit, but I have lots of people asking.   I wrote QLiteOSD as well as the lighter VoltOSD (smd21) but just wondering if someone has solved the MSP issue.   Thanks!

ramiss, 2023-12-28:
Hi.  Franky I've never tried.   This project was developed to simply get the 03 into full power state for people who don't use a flight controller.  Since there is no flight controller,  there isn't much telemetry to show.

abritten, 2023-12-29:
Hi @ramiss and @Qrome great that you both are talking about this issue! I have checked into both your Air unit repos in the past, and because I have O3, I have been using ramiss setup with the SAMD21. I did make a slight mod where I have my Teensy flight controller (DrehmFlight) send a high or low to signal to the Seeeduino to switch back to low power mode when I disarm the drone, and arm it again to high power mode. My Teensy uses LIDAR and I plan to have some barometer telemetry and hopefully GPS with modules. If you guys ever need a dedicated tester for any module inputs for O3 OSD I am willing to test and try! You can check out my very recent O3 branch with ramiss's code. https://github.com/abritten/dRehmFlight-StantonModel-VTOL/tree/DJI-O3-Air-Unit but please let me know how I can help with any collaboration and testing!

Qrome, 2023-12-29:
Thanks @abritten -- yeah, I made a simplified version of the QLiteOSD under the repo of VoltOSD over a year ago using the SAMD21 and I also use the PWM high and low to arm and disarm.   Repo here:  https://github.com/Qrome/VoltOSD  
I plan on testing some 03 Air unit with the QLiteOSD -- it also has the ability to do the arm and disarm using a PWM signal.   Feel free to reach out to me there and I would love to have additional people test the changes.  Thanks.

TodWulff, 2023-12-29:
DJI 's implementation of the O3's MSP-based OSD is a bit underwhelming, given it is purely char based without a means to load custom fonts/icons.

My current use case is on several ardupilot based birds and it is a less-than-intuitive path to traverse to get it working in a manner only peripherally close to what others are achieving with non-DJI implementations.

I used this tool on my first endeavors into FPV (a non-FC construct).

I'm looking forward to alternatives that would enable custom sprites/icons and other customizable OSD elements (fonts/etc) beyond what the DJI FW supports, if practical.

Sadly, I suspect that achieving same sans OEM support is a bridge too far.

But, I'm still lurking and will remain hopeful.  Keep up the good work, folks.

~MHz

ramiss, 2023-12-29:
I have great admiration for anyone coding their own flight controller, but my personal feeling is that anything that needs features more than this solution should probably go with an off the shelf FC with one of the very mature firmwares such as iNav, Ardupilot or Betaflight.  The cost is minimal for the massive jump in features.

closed 2024-01-05 by ramiss