ramiss/arduino_DJI_03_RC_ARM #23: Show craft name or other elements?

mf0o, 2023-12-28:
Hi Richard.

I'm currently tinkering with your code and I'm able to arm my O3 after a few seconds but it wont show any other elements like the craft name or other elements.

Is this still supported in the current state?

thank you in advance.

ramiss, 2023-12-28:
Hi. This project never supported custom OSD elements.  The only reason you can see the voltage is because that element is built into the DJI display and the code just makes sure the data is transmitted.

closed 2024-01-05 by ramiss

hichol, 2024-12-06:
this is not issue I just want to share what I found.
I just got my first DJI O3 for my Rover project, it seem this project was what I need, but then this Air unit was don so hot, I add fan to it and still hot to touch so I want to see the temperature  in DJI goggle 2, in Betaflight there is OSD element call VTX_TEMP but neither air unit or goggle didn't display anything about VTX_TEMP. So I start poking and digging for a week and this is what find it
DJI O3AU and Goggle 2 has another ways to display, I couldn't find any correlation with osd config with position and element IDs, but I left alone original code and this air unit and goggle 2 are using so call MSP_DISPLAYPORT where i just add some code 
in function send_msp_to_airunit() add this lines after send osd config

  sprintf((char *)Tdata3,"%c%c%c%cVTX TEMP:%0.1fC",3,19,0,0,getTemp());
  //Serial.printf("VTX TEMP:%0.1fC",getTemp());
  msp.send2(MSP_DISPLAYPORT, Tdata1, sizeof(Tdata1));
  msp.send2(MSP_DISPLAYPORT, Tdata2, sizeof(Tdata2));
  msp.send2(MSP_DISPLAYPORT, Tdata3, sizeof(Tdata3));
  msp.send2(MSP_DISPLAYPORT, Tdata17, sizeof(Tdata17));

at the top where defines and decollations section are add this arrays
// O3 AU OSD GRID 20 X 40 MSP DISPLAYPORT format  |catagory Byte| Y position | X position| ID Byte | Data
uint8_t Tdata1[] {0};  //you have to add these two line start sending.
uint8_t Tdata2[] {2};
uint8_t Tdata3[] {3, 0, 0, 0, 'T', 'R', 'X', '4'}; // you may add many as you want on display long as different location limit 15 
              .
              .                                                 characters
              .
uint8_t Tdata17[] {4}; // this line or array to terminate I think.

I don't know  MSP_DISPLAYPORT 182 is element ID or not it needs to in array.
as you see you need new send function but original send function exactly same except this lin of code  _stream->write('>');
make another send2() copy and paste change name of function and replace code.
and of course you make changes to hardware and add another ADC input to measure temperature.

it seems most elements are predetermined values so I can't not or don't know how to change it, but you have 15 charecters to play with that just fine with me .
I hope this would help.