ramiss/arduino_DJI_03_RC_ARM #24: Compilation error: MSP.h: No such file or directory

simranjeetsingh21, 2024-01-05:
Hello Ramiss,
I am new in this and I am just following the instructions given buy you on RC arm.
there is an error and I have no idea how to solve this.

C:\Users\simranjeet singh\Desktop\arduino_DJI_03_RC_ARM-main\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:28:10: fatal error: MSP.h: No such file or directory
 #include "MSP.h"
          ^~~~~~~
compilation terminated.

exit status 1

Compilation error: MSP.h: No such file or directory

simranjeetsingh21, 2024-01-05:
I did same but still give me the same error

ramiss, 2024-01-05:
Did you keep everything on the exact same file structure?

simranjeetsingh21, 2024-01-05:
exactly same file structure

simranjeetsingh21, 2024-01-05:
is there some thing else can cause this error or just a MSP file ?

simranjeetsingh21, 2024-01-05:
is it possible to install MSP file separately?

ramiss, 2024-01-05:
You must not.  See step 2. This version is a custom version

ramiss, 2024-01-05:
Send me a screenshot of your folder structure

simranjeetsingh21, 2024-01-05:
[image]
[image]
[image]
[image]

simranjeetsingh21, 2024-01-05:
did I send you the right things that you want to see?

ramiss, 2024-01-05:
No.  I'm talking about the folder structure on the computer.  All of the files need to be in the same folder, like you see them in the GitHub repository.

simranjeetsingh21, 2024-01-05:
[image]

simranjeetsingh21, 2024-01-05:
one more question when I choose to upload files the only first file " arduino_DJI-O3_RC_ARM" is shown in the folder not the rest of others you can see in the attached photo. Is it supposed to be like that?
[image]

ramiss, 2024-01-05:
I see your issue.  
1. Create a folder on your computer called "arduino_DJI_03_RC_ARM"   exactly that name, no "-main" or anything else.
2. Then put all of the files into that folder.
3. Open the arduino_DJI_03_RC_ARM.ino file from that folder.

simranjeetsingh21, 2024-01-05:
you are the great MAN, finally it is uploaded
so now if I connect it with DJI O3 air unit it will automatically goes to full power mode or there are some other steps?

ramiss, 2024-01-05:
Yes,  after two seconds of power the air unit will arm (if wired correctly).

simranjeetsingh21, 2024-01-05:
I hope i ll give you good news

dennisjules, 2024-01-19:
Hi Ramiss, 
First of all, thanks a lot for sharing this and to those who contributed, it worked perfectly. 
Curious if you have steps to arm O3 via flight controller coz im planning to setup with gps and osd.
Thanks again.

ramiss, 2024-01-19:
> Hi Ramiss, First of all, thanks a lot for sharing this and to those who contributed, it worked perfectly. Curious if you have steps to arm O3 via flight controller coz im planning to setup with gps and osd. Thanks again.

Hello.  Glad it worked for you.  That said,  if you use a flight controller then the Arduino becomes redundant,  in fact you shouldn't use them together.  The flight controller will send the arm command and the OSD from the GPS etc.    This android project was only devised for those that didn't want to bother with a flight controller.

abritten, 2024-01-19:
Hi @ramiss thanks again for your efforts. Regarding a general flight controller arming, or your SAMD21 arming the O3, I currently use both because my flight controller is actually a Teensy arduino and is using a open source code DrehmFlight. So I have your SAMD21 arming the O3 becasue my Teensy is not capable of doing it (lacking code). However because the Teensy Drehmflight is written in Arduino, I was hoping to copy your code and try to get the Teesny to replace the SAMD21 and arm the O3 solely by itself. But I am not a solid programmer, nor do I fully understand the structure of your MSP files. Would this be a project you would like to work on with me? So it would be a similar repo as this SAMD21 version, but the Teensy version. Drehmflight is becoming more popular with the hobbyist drone community because of its VTOL simplicity and customizing capabilities that Betaflight and Ardupilot cannot accomplish. I think getting the O3 hacked with the Teensy Drehmflight would be an awesome contribution if you want to help make it happen!

ramiss, 2024-01-19:
Hi @abritten - I'm honored to be asked to contribute, but I simply don't have the time.  You would only need to incorporate the code within my .ino file into your own code.  All of the other files can simply be included.

dennisjules, 2024-01-20:
> The 03 automatically goes into full power mode once it sees the arm signal from the flight controller (which this arduino project simply mimics). For a flight controller, you hook up the RX/TX (or S-BUS) to the flight controller and then configure the flight controller so that it knows the 03 is on that port. You then tell the flight controller to arm using a switch on your transmitter. How you do all of that depends entirely on which firmware you are using (betaflight, iNav, Ardupilot etc). Without any more info I can't offer any more instruction.

@ramiss Currently I’m using below setup
HGLRC Zeus F722 flight controller and using betaflight 4.4, I did not setup any AUX under MODES section in BF.
Radiolink RC6GS radio
Radiolink R7FG receiver with gyro enabled
Dji O3 unit
Dji goggles 2 with low power mode disabled
BKRC head tracker
Everything is working fine and in my goggles I see the OSD I needed and 50mbps on my video signal but the moment I turn it on, it is not auto recording compared with your arduino setup.
My question is, with that 50mbps video signal is it right to assume that my O3 unit is in arm state without setting anything inside betaflight?  If the answer is no, how can I set it up in BF to arm my O3 but do I need to arm considering the 50mbps?
Because from youtuber mads tech that 50mbps doesn’t mean the O3 is armed just by disabling low power state, so we cannot use the full power of O3 unless we send an arm command to O3. So this is what I’m trying to find out. Thanks.

ramiss, 2024-01-21:
@dennisjules if you see the custom OSD and 50mbps then the 03 is at full power.  You either need to enable the record on arm setting and/or your SD card is full.

dennisjules, 2024-01-21:
> @dennisjules if you see the custom OSD and 50mbps then the 03 is at full power. You either need to enable the record on arm setting and/or your SD card is full.

That’s what I thought but I compared the arduino versus my FC setup. Arduino range is farther so that means my FC setup although seeing 50mbps was not indeed using it’s full power since it is not arm as mads tech mentioned in his youtube video. So finding ways now on how to arm O3 unit under FC setup.

abritten, 2024-01-23:
> That’s what I thought but I compared the arduino versus my FC setup. Arduino range is farther so that means my FC setup although seeing 50mbps was not indeed using it’s full power since it is not [...]

I tested the power consumption for the 03 air unit on a power supply, arming it via a Betaflight FC and then SAMD21. Both had equal power consumption:

Standby = 4.5W
Armed = ~11W

dennisjules, 2024-01-23:
> I tested the power consumption for the 03 air unit on a power supply, arming it via a Betaflight FC and then SAMD21. Both had equal power consumption:
>
>
>
> Standby = 4.5W
>
> Armed = ~11W

What you mean by power consumption? The RF output?

borisbl4, 2024-01-25:
[image]
[image]

Hello, I'm trying to upgrade to a different version of the board.  I don’t quite understand this. There is a mistake.

[image]

D:\Загрузки\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino: In function 'void setup()':
D:\Загрузки\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:87:3: error: 'analogReadResolution' was not declared in this scope; did you mean 'analogWriteResolution'?
D:\Загрузки\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino: In function 'void debugPrint()':
D:\Загрузки\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:334:3: error: 'SerialUSB' was not declared in this scope; did you mean 'Serial1'?

exit status 1

Compilation error: 'analogReadResolution' was not declared in this scope; did you mean 'analogWriteResolution'?

ramiss, 2024-01-25:
Both of those functions are depreciated in later board libraries, which is why the Seeeduino instructions specify to use an older board "version".   You could try try lowering the board version for your board or investigate the replacement code for those functions.  You can safely comment out the SerialUSB line as that just spits out debug code.  You might try also commenting out the analogReadResolution - it may not even be needed on your board.

dennisjules, 2024-01-27:
@ramiss Tested again today
Using FC at 50mbps but video signal drop at 1 bar at 100meters
Using arduino no video signal drop at 350meters
Therefore I can confirmed using FC doesn't arm O3 unit eventhough you disable low power mode and at 50mbps.

ramiss, 2024-01-27:
Appologies, but without more information on your setup I can't diagmose it. Also,  just pointing out that you are asking in literally the opposite location.  You should find the forum for that flight controller or firmware ;)

dennisjules, 2024-01-27:
@ramiss no worries, I'm just sharing my experience in case somebody reads this thread. There was a lot of confusion from other threads that using an FC will arm the O3 unit when we disabled the low power mode but it's not the case. As mentioned before, I'm looking now for an auto arm feature in Betaflight. For now the only way I can do that is to use my dji fpv radio controller. Just curious though, can you share which portion in the arduino code that arms the O3. Thanks 🙏🏼

ramiss, 2024-01-27:
> can you share which portion in the arduino code that arms the O3. Thanks 🙏🏼

There's not a whole lot of fluff in the code.   The arm command is below, but that it pulling configuration data from the other variables and structures and sending through the MSP library.  

```
//MSP_STATUS
  status_DJI.flightModeFlags = flightModeFlags;
  status_DJI.armingFlags = 0x0303;
  msp.send(MSP_STATUS_EX, &status_DJI, sizeof(status_DJI));
  status_DJI.armingFlags = 0x0000;
  msp.send(MSP_STATUS, &status_DJI, sizeof(status_DJI));
```

ismail3887, 2024-01-29:
Hello, I do as you explained in the article and diagram, but I still get errors. Can you make a short video about this topic? Or can someone who has done this help me on this issue? I've been trying for 2 days but I can't do it.

ramiss, 2024-01-29:
Please post your errors.

ismail3887, 2024-01-30:
> Please post your errors. 

Doesn't it work when the power input to seediuno is supplied with 5v BEC? I didn't understand the seeediuno (any receive channel for power) part. Do I need to connect it to the + and - part of any input channel on the receiver?

dennisjules, 2024-01-30:
@ismail3887 
Seeeduino can be powered by any power source as long as it is 5v nothing more. I powered my seeeduino from my receiver channel and everything works great.

ismail3887, 2024-01-30:
> Seeeduino can be powered by any power source as long as it is DC 5v nothing more. I powered my seeeduino from my receiver channel and everything works great.

Did you do it like in the photo? Can we please talk on a platform that will communicate with me faster? With Instagram or any five social media. I need to solve it as soon as possible

dennisjules, 2024-01-30:
[image]

ismail3887, 2024-01-31:
> [image]

Can you write me on Instagram? ( @ismail_fpv )

ismail3887, 2024-01-31:
> Please post your errors. 

Did I manage to install the software?[image]

ramiss, 2024-01-31:
@ismail3887 - You literally posted your answer.  Software did not install, follow the directions under possible causes.  Also, make sure you are set to the proper serial port.  If you are still having issues then go watch other tutorials on the basics of Arduinos.

ismail3887, 2024-01-31:
> @ismail3887 - You literally posted your answer.  Software did not install, follow the directions under possible causes.  Also, make sure you are set to the proper serial port.  If you are still having issues then go watch other tutorials on the basics of Arduinos.

Is the correct serial port as seen in the photo? Is everything correct?

ramiss, 2024-01-31:
@ismail3887. It depends on your computer setup.  Please read about arduino basics and then try again. 
https://www.digikey.com/en/maker/tutorials/2018/how-to-get-started-with-arduino

closed 2024-03-08 by ramiss