ramiss/arduino_DJI_03_RC_ARM #28: Can’t seem to get correct voltage on osd.

droplow77, 2024-03-06:


droplow77, 2024-03-07:
Also getting error “incompatible goggles, air unit, receiver” message in the goggles after about 30 seconds being powered up

ramiss, 2024-03-08:
@droplow77 - For the voltage - it is not so important that the resistors are exact, but much more important that the ratio between the values is close to what is listed in the readme file.  It is also possible that your resistors are not of high quality and you might change them.

For the incompatibility, this is due to the goggles and air unit not being on the same firmware versions.

droplow77, 2024-03-09:
Thanks for the reply.  

I used the exact values (ordered on Amazon) that you listed.  Possibly low quality.  ¯\_(ツ)_/¯ I also tried different resistors previously, before my Amazon resistors arrived, by combining ones I had on hand in series/parallel to create the proper resistance value. 

If I was successful at getting the values within the correct ratio, do I need to adjust the value of VOLT_DIVIDER to get the osd voltage to correctly match the lipo?

In your testing, does your osd voltage fluctuate up and down within a 2v range, like mine? Or is it constant and holding?

ramiss, 2024-03-09:
If you are getting fluctuations without load then something fishy is going on. This is the most basic way to sense voltage and not perfect, but you should be able to keep a stable value.  If you really want accuracy, you should pull your cruise throttle load and then calibrate the volt_divider value to actuals.

droplow77, 2024-03-09:
Yes, I am getting fluctuating value in the osd without load.   It is fluctuating between .7 and 2.1 volts, approximately.  

What should my approximate voltage reading be at the A0 pad be if I have a charged 4S battery?   

I assume, based on your calculations, 16.8 / 6.471 = 2.59, but, I want to verify my assumption, so I can troubleshoot my voltage divider.  

I assume it needs to be within the range of 0-3.3v???

Thanks again for your help.

ramiss, 2024-03-09:
You can use a voltage divider calculator like this:
[https://www.digikey.com/en/resources/conversion-calculators/conversion-calculator-voltage-divider](https://www.digikey.com/en/resources/conversion-calculators/conversion-calculator-voltage-divider)

I've found them to be relatively accurate.  One thought is that you somehow did go over 3.3 volts and damaged that pin.  You might try another pin and yes, keep your max potential voltage under 3.3.

droplow77, 2024-03-09:
Thanks for your answer. 
I checked my voltage regulator with a multimeter and it is showing a stable voltage just above 2v with a half charged 4S. 

I will try another pin next.  Thanks for the advice and clarification.

droplow77, 2024-03-09:
I think I damaged my A0 pin as you suggested.   

Using A1, I got more stable voltage, only fluctuating by approximately 0.4v. 

Thank You for your input.  

Now, I need to adjust my VOLT_DIVIDER  value.  

Thanks again

ramiss, 2024-03-09:
Nice troubleshooting.  Glad you got it figured out!

closed 2024-03-09 by ramiss