ramiss/arduino_DJI_03_RC_ARM #30: Thanks for this but how do I get sbus working?

jjziets, 2024-05-17:
Hi. Thanks you alot for this work. I can now use my air 03 unit for other uses cases. Just one thing that I wish I can do is get the controller signal from the unit. I have dji remote control but the unit is not providing any data on the sbus. I assume I need to tell it to do so?

Any idea how or where I find that information?

TodWulff, 2024-05-17:
On my O3, Sbus just worked.  Bound the goggles to the O3, bound the TX. Got a video link and Sbus data appears on the yellow wire coming from the O3.

jjziets, 2024-05-17:
I did not know I should bind the remote control to the air unit. I assume that is what you me with TX unit.

TodWulff, 2024-05-18:
Yeah, the remote control (tx) needs to be bound to the O3. See the Linking section from here: https://support.dji.com/help/content?customId=en-us03400007099&spaceId=34&re=US&lang=en

ramiss, 2024-05-18:
@jjziets - Hopefully you are figuring out what you need.  I'm a little confused by what you are trying to do though.  The arduino_DJI_03_RC_ARM project is designed to arm the 03 in the absence of a flight controller. Using s-bus along side it implies you are using a flight controller.

jjziets, 2024-05-18:
Yes thanks for the code. I had to port it to c as I'm suing a STM32 but I
want to control another platform with the dji Air 03 unit. I have an DJI
Avana and did not realize I had to bind the remote with the air unit again.
I thought it was bind with en Googles.

Is there any other way to contact you? I'm Etherion on discord and I'm on
Betaflight server

ramiss, 2024-05-18:
> Yes thanks for the code. I had to port it to c as I'm suing a STM32 but I want to control another platform with the dji Air 03 unit. I have an DJI Avana and did not realize I had to bind the remote [...]

The Arduino_DJI_03_RC_ARM code pretends to be a flight controller talking TO the Air Unit over the UART pins.  I've never tried, and I'm not sure if anyone has tried, also connecting a second platform (ie flight controller) to sbus, while also having a flight controller on the UART.  Good luck trying and let me know, but I don't really have any experience to offer.

jjziets, 2024-05-20:
Yes thanks. It seems the Air O3 Unit doesn't work with the DJI motion controller 2 that I got with the Avata, which negates  the reason why I wanted to get the Air O3. But I am now so far down the rabbit hole that I might as well continue.

the OSD code on your git does it work with the DJI air o3 unit? I can set the bat voltage, and I can bring it out of low power mode now, but I don't see anything else on the google 2 that I got with the avatar. I would like to show power and speed, maybe even directions.  Maybe I port the code wrong so just need to know if there is hope

ramiss, 2024-05-20:
How do you expect to get power, speed,  direction etc without a flight controller? There is no OSD to display when using this project.

jjziets, 2024-05-20:
Well there is the osd configuration and setup as well as the osd send
function. The microcontroller is acting as a fc. It's basically what you
are doing. I will check how betaflight does this.

ramiss, 2024-05-20:
You misunderstand me.  Yes, it is possible to have the arduino provide OSD information to the air unit.  What I'm saying is that you don't have that information available unless you also add sensors to the arduino and, at that point, why not just use a flight controller?  This project is designed for people who don't want to bother with a flight controller and just want video transmission at full power.

jjziets, 2024-05-20:
Sure. I'm not using it in a drone so don't need al the Flight controller
math. The RC car has speed and direction and battery. So I want to send
that and throttle.

Right now I got the osd sending but nothing shows on the screen no dummy
values other than battery voltage.

closed 2024-06-21 by ramiss