ramiss/arduino_DJI_03_RC_ARM #32: Receive Control Data from O3 Air Unit

Vinz1911, 2024-05-24:
Hi,

I'm currently looking for a good long range and reliable remote control solution for my robotics project. I find this repo and im interested in if it's possible to read the data from the O3 Air Unit via UART. My idea was to connect an FPV Controller 2 to the O3 Air Unit (Without goggles) and read the necessary data from O3 Air Unit. By developing this library did you tested something in that way? is it possible to receive the necessary control data from the O3 Air Unit for a possible usage in other Arduino based Projects?

Thank you,
Greetings Vinz

ramiss, 2024-05-24:
Yes,  the DJI air unit communicates over MSP. 
https://github.com/yajo10/MSP-Arduino

Vinz1911, 2024-05-24:
Thanks for your fast reply. Then i think i can Build an appropriate solution for my own needs. 😊

ramiss, 2024-05-24:
@Vinz1911 - in case you aren't aware, there are plenty of flight controllers out there that will do full MSP with the air unit and control servos for almost any vehicle.

Vinz1911, 2024-05-25:
Thanks but I flight controller would be an additional component that I don't really need. So it would be great just to read the data from the FPV Controller. 

I tried today and connected the O3 Air Unit over UART to the Arduino I have, I used your library and I also see that the O3 Unit gets armed. I tried with this code to get the RC Channels to read the data from the FPV Controller 2 

```c++
void debugPrint() {
    msp_rc_t rc;
    if (msp.request(MSP_RC, &rc, sizeof(rc))) {
      for (int i = 0; i < MSP_MAX_SUPPORTED_CHANNELS; i++) {
        Serial.print("Channel ");
        Serial.print(i);
        Serial.print(": ");
        Serial.println(rc.channelValue[i]);
      }
  }
}
```

The FPV Controller is connected to the O3 Air unit, but for every channel it only prints 0. don't see any changes by moving the sticks or pressing buttons. Did I missed something here? Do I need to send any command before?

Thanks

Vinz1911, 2024-05-26:
FYI: 

it seems that it's not possible to get the FPV Controller data over UART. This data will be provided by the SBUS interface of the O3 Air Unit. I did a quick solder today and build an inverter an now I'm able to retrieve the necessary channel information. 

Greetings

ramiss, 2024-06-21:
Glad to see you got your issue resolved.  The path you went on is beyond the scope of this project, but good info for others.

closed 2024-06-21 by ramiss