ramiss/arduino_DJI_03_RC_ARM #33: Low power mode

GrumpyLT, 2024-05-31:
I see the behavior mbs jump to 50 as described.  In your work on this, do you know why the air unit in low power mode banner stays on tho?  Do you know what controls that portion?  Is that normal?  I know I can turn that off in the settings, so I assume as long as I see the mbs jump, the device is working properly?

TodWulff, 2024-05-31:
Apologies for jumping in the middle here.

A UK gent by the name of Ian (YouTube: Mad's Tech, iirc) had, a couple of years ago, identified that there are actually 3 operating modes of the O3: low power, high power & armed, and some hybrid of the two which presents the higher throughput but in a low RF power context, iirc.

I recommend that you take a peek at his related videos to attempt to gain a higher fidelity understanding.

Just my $0.03.  YMMV.  Good luck.

TodWulff, 2024-05-31:
Also, I've not observed the behavior I perceive you to be describing (50mbps throughput but low power banner still depicted on the DJI OSD). it's been my experience that when the Xiao successfully arms the O3, it immediately jumps to 50mbps, the low power banner clears, and, if so configured, starts recording on the air unit and goggles. 

Just to affirm, I've not had the O3 configured to avoid low power mode with any of my setups.

I use this project's code in several contexts: rc car use, Fixed-Wing use w/o a Flight Controller, and in a hybrid context with my Xiao Sense based Roll-o-Meter (that project is here on Github).

This project's sketch has been robust and 'just worked' for me in numerous instances of these contexts.  I'd verify that your O3 is updated to the latest FW too.

ramiss, 2024-06-21:
> I see the behavior mbs jump to 50 as described. In your work on this, do you know why the air unit in low power mode banner stays on tho? Do you know what controls that portion? Is that normal? I [...]

Yes, as long as you see the Mbps jump up then this project is doing its thing.  Not sure why you are seeing the additional banner though. Latest DJI firmware?

closed 2024-06-21 by ramiss

GrumpyLT, 2024-06-21:
Ramiss,

I've been playing with the old nano code for a bit and here is what I found.

When using aux channel to arm, the low power banner goes away, but when
using the always_arm variable it does not.  This peaked my interest a bit,
so I played with the code.  If you send a De-Arm signal first, then an arm
signal, the low power banner goes away.  But if you just send the arm
signal straight away, the low power banner does not go away. (MBs still
jump tho)  Odd, but because this is the old code, I dont think that it will
be useful at all to your project.

I finally got a Seeed in the mail, so I will be moving over to the newer
code. I suspect this won't be the case on this platform, but we will see.