ramiss/arduino_DJI_03_RC_ARM #34: Error setting up

Drvr4spd, 2024-06-20:
Hi, 1st time trying to mess with this, I've downloaded the files, but in arduino I can not find any boards named seeduino xiao at all. I've tried to use arduino micro as the board says it's equivalent to that, but when I try to download the sketch into the card I get a compilation error: 'analogReadResolution' was not declared in this scope. Any help would he greatly appreciated,  thanks 
[image]

ramiss, 2024-06-20:
The analogReadResolution line is specifically needed for the Seeduino and should be commented out for other boards.

Incidentally, here is how you install the Seeduino boards in the Arduino IDE, if needed:
[https://wiki.seeedstudio.com/Seeed_Arduino_Boards/](https://wiki.seeedstudio.com/Seeed_Arduino_Boards/)

Drvr4spd, 2024-06-20:
Thank you, I'll run through that in the morning and load the seeduino boards, hopefully finally get it all straight and run it in my f18 this weekend at full power with the head tracker

Drvr4spd, 2024-06-21:
thanks it worked perfect

closed 2024-06-21 by Drvr4spd