ramiss/arduino_DJI_03_RC_ARM #36: Screenshot of the (expected) OSD

maxpolzin, 2024-07-30:
@ramiss This is great work, thx a lot. I slightly modified your code to run on a XIAO ESP32-C3 and the DJI O3 Air Unit seems to transmit correctly to my DJI Goggles 2, https://github.com/maxpolzin/arduino_DJI_03_RC_ARM. 

If I understood correctly, there should be some information displayed inside the goggles. I think I spot the battery voltage in the bottom right. 

The flight mode should somehow be displayed all 10s, if I read the code correctly. Where would I see the flight mode inside the goggles (to be sure that I transmit in high power mode)? 
What other information should be displayed? Do I need to enable these settings somewhere in the goggles?

ramiss, 2024-07-30:
My code ignores all OSD elements because the purpose is to assist when not flying with a flight controller.  The craft voltage you see is DJI's baked in OSD element for the craft.  You'll need to work on the code to add elements.  The positions can be found in "OSD_positions_config.h".

maxpolzin, 2024-07-30:
Thank you for the immediate response and explanations.

Good to know, I guess I can remove then some unused references/code in the
main .ino file.

Did you ever manage to display any custom field in the goggles? For example
the battery level?

maxpolzin, 2024-07-31:
Oh, I think I understand it better now. The voltage is always displayed in the OSD by DJI. Though, by setting the variable in the dji_status variable in your code and sending it via msp, it actually updates the value in the goggles. Is that correct?

thibaultvdb, 2024-08-05:
@maxpolzin I'm trying to run the program on an esp32 (lilygo mini32 esp32)
Do you have any pointers on how to adjust the program to get it working on ESP instead of the arduino?

ramiss, 2024-08-05:
For the most part, arduino code will compile on an esp32.  However, there are enough subtle differences that certain libraries may not work.  The best approach is to compile for esp32 and work through the errors.  You may need to find alternative libraries or modify the ones provided.  Sorry I can't give more info, although I've ported other code to esp32, I have yet to try with this project.

maxpolzin, 2024-08-05:
@thibaultvdb My fork, https://github.com/maxpolzin/arduino_DJI_03_RC_ARM/tree/mx/cleanup, compiles and works nicely on a Seeed Studio XIAO ESP32C3.  Maybe this helps you as a starting point to compile on your ESP32 platform. The main difference is on the ESP32's ability to allocate UART ports dynamically to output pins. Let me know if you need further help.

thibaultvdb, 2024-08-05:
Thanks for the pointers! I have the XIAO on the way but I'm going to give it a try anyhow 👍

closed 2024-11-01 by ramiss