ramiss/arduino_DJI_03_RC_ARM #38: cannot verify or upload ino file

66xing91, 2024-10-28:
C:\Users\javier\Documents\Arduino\libraries\arduino_DJI_03_RC_ARM\arduino_DJI_03_RC_ARM.ino:28:10: fatal error: MSP.h: No such file or directory
 #include "MSP.h"
          ^~~~~~~
compilation terminated.
exit status 1

Compilation error: MSP.h: No such file or directory

ramiss, 2024-10-28:
You need to download the entire repository.  Not just the 1 file.

66xing91, 2024-10-29:
Good afternoon thank you for taking the time to reply back to me, I was
able to install the rest, I didn't do what you said, even after reading the
instructions. I know I'm an idiot. One more question how do I know when the
unit goes full power?

ramiss, 2024-11-01:
You will know when the low power message in the OSD (bottom right orange message) goes away and the bitrate jumps up to the expected settings (25mbps or 50mbps).  If the mbps stays in the single digits then it is not armed.

closed 2024-11-01 by ramiss