ramiss/arduino_DJI_03_RC_ARM #39: XIAO-RP2040

magixfly, 2024-11-01:
Hi! 
i got the XIAO-RP2040
https://wiki.seeedstudio.com/XIAO-RP2040/ 

i did all the installation steps succesfully, but O3 doesnt arm. 
will i need another code or do i have to buy XIAO SAMD21 ?

ramiss, 2024-11-01:
Because it's a different board, you'll need to look at the pin assignments and adjust.

magixfly, 2024-11-01:
ok, i hope i understand. the boards have the same pin assignment. XIAO SAMD21 - is the one you use?

RP2040: D6-TX / D7-RX
SAMD21: D6-A6-TX /// D7-A7-RX

ramiss, 2024-11-01:
That's fair, they do seem to have the same pins and capability.  I assume the code compiled without error?  If so I don't have a clear explanation as, in theory, it should work.   Correct that this project was built with the Seeed Studio XIAO SAMD21(Seeeduino XIAO).  https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html

magixfly, 2024-11-01:
yes, compiled without error. i dont know much about code. but i saw some lines with "analog..." in your code ?!
I will buy SAMD21 and try next week.
thank you

closed 2024-11-02 by magixfly